chungmin99/pyroki #4: Batched example?

Kashu7100, 2025-05-08:
Thanks for your amazing work. 
Do you have any batched trajectory optimization example by any chance?

chungmin99, 2025-05-12:
Hi!

This [snippit](https://gist.github.com/chungmin99/2b62427ecc85d1c248353439ebd50424) is a batched version of the [existing trajopt example](https://github.com/chungmin99/pyroki/blob/main/examples/07_trajopt.py). 

It's very close to the original script, but instead uses JAX's built-in mapping operations (e.g., `jax.vmap`). You can find another batching example in the IK benchmarking script [here](https://github.com/chungmin99/pyroki/blob/90ca2925f108d5edc67d0d878d858bc79310daad/benchmark/ik_benchmark.py#L194)!

The output should look similar to this:

https://github.com/user-attachments/assets/36e281d1-682c-4200-89ce-c3c344166fdc

Kashu7100, 2025-05-13:
Thanks! I will play with it

chungmin99, 2025-05-18:
Closing as resolved for now -- feel free to re-open if you have further questions!

closed 2025-05-18 by chungmin99