chungmin99/pyroki #6: Singularity Handling

peasant98, 2025-05-08:
Hi all, 

Thanks for the great work!

I haven't fully gone through the code, but I was wondering if there is an opportunity to integrate singularity *handling* in this framework, where we can safely enter and exit singularity.

Some folks and I recently got out a work called [J-PARSE](https://jparse-manip.github.io/) that could fit into this work, what do you think?

Splash pic 
[image]

chungmin99, 2025-05-19:
Hi -- this is super cool work!

The code includes a simple manipulability cost implementation ([here](https://github.com/chungmin99/pyroki/blob/769c3f78d93934285d08a538dbcfc2010768f5de/src/pyroki/costs/_costs.py#L147)), and if I understand correctly it should be easy to write a version of this cost where J-PARSE's safety jacobian is used instead.

Happy to take contributions in the examples :-)

peasant98, 2025-06-02:
Great! I will integrate our method in the repo this or next week.

peasant98, 2025-06-13:
Nice @amjack0, I have created a fork of this repo (https://github.com/peasant98/pyroki) and will add to it today+tomorrow.

chungmin99, 2025-07-18:
Hi -- sorry for getting back to this so late, I've been away for the summer 🥲 ...

I just checked the fork but didn’t see any new commits, so I wanted to double-check if I might’ve missed anything!

peasant98, 2026-01-21:
I have returned from the cave

We tried an example [here](https://github.com/armlabstanford/jparse/tree/main?tab=readme-ov-file#pyroki-integration-3d-interactive-visualization), where we follow the exact J-PARSE implementation with pyroki!! Maybe we could have this for examples? I'll iron out a few bugs as well.

Video
https://github.com/user-attachments/assets/9990f104-e5ee-411a-8a85-8a81c6346c7c