chungmin99/pyroki #8: Online Planning for Dual-Arm Humanoid Robots: Missing Self-Collision Detection and Target Link Configuration Errors

kelejichi6463, 2025-05-12:
Great job! But there are two issues: the collision detection doesn't seem to account for the robot itself; with a dual-arm humanoid robot, collisions between the arms and the robot body are not effective in online planning. Secondly, in online planning, if you add target_link_names = ['yumi_link_7_r', 'yumi_link_7_l'], it throws an error.

chungmin99, 2025-05-18:
Hi!

> the collision detection doesn't seem to account for the robot itself

I did just find a bug in the pairwise collision checking code (see #15), this should fix the problem with the self-collision you were seeing! You should be able to test it with a collision-aware bimanual IK example [here](https://gist.github.com/chungmin99/775c3ce1a0c21caa76d80a166006cab7), merging two existing examples.

>  with a dual-arm humanoid robot, ... are not effective in online planning.
> Secondly, in online planning, if you add target_link_names = ['yumi_link_7_r', 'yumi_link_7_l'], it throws an error.

The current online planning example only supports one joint target for the panda, is this based on a custom script?

https://github.com/chungmin99/pyroki/blob/d9228a00a07b5c8fbe5262a9b3c552ec40290c72/examples/pyroki_snippets/_online_planning.py#L12-L24

(`target_link_name` expects a single string.)

Also, I'm not sure what you mean by online planning for dual-arm humanoid, it seems that you're referring to the YuMi in `examples/02_bimanual_ik.py`?

Either way, a short reproducible snippit or a copy of the error would be helpful!

kelejichi6463, 2025-05-23:
Thank you, the latest example you provided successfully resolved the collision issue.

chungmin99, 2025-05-23:
Great -- if you don't have any other questions / issues I'll close this one!

closed 2025-06-06 by chungmin99