chungmin99/pyroki #9: mis‑aggregated Jacobian columns when passive joints precede actuated joints or when mimic joints are present

lsun-bdai, 2025-05-13:
#### Description
`_pose_cost_analytic_jac._pose_cost_jac` currently collapses the full  `num_joints` Jacobian to  `num_actuated` like [this](https://github.com/chungmin99/pyroki/blob/90ca2925f108d5edc67d0d878d858bc79310daad/src/pyroki/costs/_pose_cost_analytic_jac.py#L157)

```python
jac = jac[:, robot.joints.actuated_indices][:, : robot.joints.num_actuated_joints]
```

I find this during my test on my customized franka urdf, with a extra fixed base joint and extra actuators and fixed joints between joint 7 and hand, so the actuated_indices becomes  `[-1, 0, 1, 2, 3, 4, 5, 6, -1, -1, 7, -1, -1, 8, 8]`. The IK solution then becomes weird.

After some searching I found this line of code might be the problem. I also saw your comments...not sure if this is the only related code.

#### Potential Fix
mask → gather → scatter‑add
```python
# Fixed‑size list of kinematic joints that are actually actuated
kin_idx = jnp.nonzero(robot.joints.actuated_indices != -1,
                      size=robot.joints.actuated_indices.size,
                      fill_value=0)[0] # (num_joints,)

# Map each kept joint → its actuator id (may repeat for mimic joints)
act_idx = robot.joints.actuated_indices[kin_idx] # (num_joints,)

# Scatter‑add merges mimic joints into their single actuator column
jac = (
    jnp.zeros((6, robot.joints.num_actuated_joints), jac.dtype)
      .at[:, act_idx]
      .add(jac[:, kin_idx])
)
```

chungmin99, 2025-05-13:
Hi yes -- we're working on a fix at #11 ! 

It's similar to the PR you made at #10, but also patches some other assumptions (e.g., that the joint index list is topologically sorted with respect to the kinematic dependencies). Let us know if this doesn't fix the problem!

lsun-bdai, 2025-05-13:
Ah yes, I didn't check the branches and didn't realize that, thanks for the fix!!!
Regarding the topologically sorting problem, I first actually solved the problem by editing the URDFs before I found what's wrong in the code. This might also related to some weird whole body IK solutions I observed in spot loco-manipulation cases, will come back if the fix isn't solving them all.

closed 2025-05-13 by lsun-bdai