chungmin99/pyroki #12: Something wrong when running with GPU?

ExplorerFzw, 2025-05-15:
Dear author, 
it is worked with cpu, but I got something wrong with gpu.
The conda environment is :
///
#
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       2_gnu    conda-forge
anyio                     4.9.0                    pypi_0    pypi
attrs                     25.3.0                   pypi_0    pypi
bzip2                     1.0.8                h68df207_7    conda-forge
ca-certificates           2025.4.26            hbd8a1cb_0    conda-forge
certifi                   2025.4.26                pypi_0    pypi
charset-normalizer        3.4.2                    pypi_0    pypi
colorlog                  6.9.0                    pypi_0    pypi
docstring-parser          0.16                     pypi_0    pypi
gitdb                     4.0.12                   pypi_0    pypi
gitpython                 3.1.44                   pypi_0    pypi
h11                       0.16.0                   pypi_0    pypi
httpcore                  1.0.9                    pypi_0    pypi
httpx                     0.28.1                   pypi_0    pypi
idna                      3.10                     pypi_0    pypi
imageio                   2.37.0                   pypi_0    pypi
jax                       0.6.0                    pypi_0    pypi
jax-cuda12-pjrt           0.6.0                    pypi_0    pypi
jax-cuda12-plugin         0.6.0                    pypi_0    pypi
jax-dataclasses           1.6.2                    pypi_0    pypi
jaxlib                    0.6.0                    pypi_0    pypi
jaxlie                    1.5.0                    pypi_0    pypi
jaxls                     0.0.0                    pypi_0    pypi
jaxtyping                 0.3.2                    pypi_0    pypi
jsonschema                4.23.0                   pypi_0    pypi
jsonschema-specifications 2025.4.1                 pypi_0    pypi
lazy-loader               0.4                      pypi_0    pypi
ld_impl_linux-aarch64     2.43                 h80caac9_4    conda-forge
libexpat                  2.7.0                h5ad3122_0    conda-forge
libffi                    3.4.6                he21f813_1    conda-forge
libgcc                    15.1.0               he277a41_2    conda-forge
libgcc-ng                 15.1.0               he9431aa_2    conda-forge
libgomp                   15.1.0               he277a41_2    conda-forge
liblzma                   5.8.1                h86ecc28_1    conda-forge
libnsl                    2.0.1                h31becfc_0    conda-forge
libsqlite                 3.49.2               h5eb1b54_0    conda-forge
libuuid                   2.38.1               hb4cce97_0    conda-forge
libxcrypt                 4.4.36               h31becfc_1    conda-forge
libzlib                   1.3.1                h86ecc28_2    conda-forge
loguru                    0.7.3                    pypi_0    pypi
lxml                      5.4.0                    pypi_0    pypi
manifold3d                3.0.1                    pypi_0    pypi
mapbox-earcut             1.0.3                    pypi_0    pypi
markdown-it-py            3.0.0                    pypi_0    pypi
mdurl                     0.1.2                    pypi_0    pypi
ml-dtypes                 0.5.1                    pypi_0    pypi
msgspec                   0.19.0                   pypi_0    pypi
ncurses                   6.5                  ha32ae93_3    conda-forge
networkx                  3.4.2                    pypi_0    pypi
nodeenv                   1.9.1                    pypi_0    pypi
numpy                     2.2.5                    pypi_0    pypi
nvidia-cublas-cu12        12.9.0.13                pypi_0    pypi
nvidia-cuda-cupti-cu12    12.9.19                  pypi_0    pypi
nvidia-cuda-nvcc-cu12     12.9.41                  pypi_0    pypi
nvidia-cuda-runtime-cu12  12.9.37                  pypi_0    pypi
nvidia-cudnn-cu12         9.10.0.56                pypi_0    pypi
nvidia-cufft-cu12         11.4.0.6                 pypi_0    pypi
nvidia-cusolver-cu12      11.7.4.40                pypi_0    pypi
nvidia-cusparse-cu12      12.5.9.5                 pypi_0    pypi
nvidia-nccl-cu12          2.26.5                   pypi_0    pypi
nvidia-nvjitlink-cu12     12.9.41                  pypi_0    pypi
openssl                   3.5.0                hd08dc88_1    conda-forge
opt-einsum                3.4.0                    pypi_0    pypi
packaging                 25.0                     pypi_0    pypi
[... 84 lines cut ...]
  File "/xxxxxxxx/miniconda3/envs/py312_gpu/lib/python3.12/site-packages/jax/_src/interpreters/pxla.py", line 3004, in from_hlo
    xla_executable = _cached_compilation(
                     ^^^^^^^^^^^^^^^^^^^^
  File "/xxxxxxxx/miniconda3/envs/py312_gpu/lib/python3.12/site-packages/jax/_src/interpreters/pxla.py", line 2795, in _cached_compilation
    xla_executable = compiler.compile_or_get_cached(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/xxxxxxxx/miniconda3/envs/py312_gpu/lib/python3.12/site-packages/jax/_src/compiler.py", line 432, in compile_or_get_cached
    return _compile_and_write_cache(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/xxxxxxxx/miniconda3/envs/py312_gpu/lib/python3.12/site-packages/jax/_src/compiler.py", line 694, in _compile_and_write_cache
    executable = backend_compile(
                 ^^^^^^^^^^^^^^^^
  File "/xxxxxxxx/miniconda3/envs/py312_gpu/lib/python3.12/site-packages/jax/_src/profiler.py", line 334, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/xxxxxxxx/miniconda3/envs/py312_gpu/lib/python3.12/site-packages/jax/_src/compiler.py", line 330, in backend_compile
    raise e
  File "/xxxxxxxx/miniconda3/envs/py312_gpu/lib/python3.12/site-packages/jax/_src/compiler.py", line 324, in backend_compile
    return backend.compile(built_c, compile_options=options)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
jaxlib.xla_extension.XlaRuntimeError: INTERNAL: cuSolver internal error

///

I am also cannot find anything about how to run this work with GPU. Thank you for your great work!

chenyenru, 2025-05-15:
This seems related to [this issue](https://github.com/jax-ml/jax/issues/8916). 
[From Jax's documentation](https://docs.jax.dev/en/latest/gpu_memory_allocation.html#:~:text=JAX%20will%20preallocate%2075%25%20of%20the%20total%20GPU%20memory%20when%20the%20first%20JAX%20operation%20is%20run.%20Preallocating%20minimizes%20allocation%20overhead%20and%20memory%20fragmentation%2C%20but%20can%20sometimes%20cause%20out%2Dof%2Dmemory%20(OOM)%20errors), "JAX will preallocate 75% of the total GPU memory when the first JAX operation is run. Preallocating minimizes allocation overhead and memory fragmentation, but can sometimes cause out-of-memory (OOM) errors".

Can you try setting `os.environ["XLA_PYTHON_CLIENT_PREALLOCATE"] = "false"`

ExplorerFzw, 2025-05-16:
I tried following your suggestion, unfortunately, it didn't work and the results were consistent with before. However, it is indeed possible to eliminate 75% of the pre allocated GPU memory, but it seems that the bug is not caused by this.

ExplorerFzw, 2025-05-16:
Furthermore, in my understanding, the current issue may be caused by compatibility issues between jaxls and cusolver or some other problem.

chenyenru, 2025-05-16:
I see, sad that didn't help. I hope your issue will be resolved soon

From reading the error status, it seems like there could be a compatibility issue with CUDA driver. ([see here](https://docs.nvidia.com/cuda/archive/12.0.0/cusolver/index.html#:~:text=CUSOLVER_STATUS_INTERNAL_ERROR,the%20routine%E2%80%99s%20completion.)) 

What does it show when you run `nvcc --version`?

Could you also send how you installed jax? It might be helpful to try `pip install --upgrade "jax[cuda12]"`.

Another thing to check would be your `LD_LIBRARY_PATH` (see [jax's instruction on that](https://docs.jax.dev/en/latest/installation.html#nvidia-gpu:~:text=JAX%20uses%20LD_LIBRARY_PATH%20to%20find%20CUDA%20libraries%20and%20PATH%20to%20find%20binaries%20(ptxas%2C%20nvlink).%20Please%20make%20sure%20that%20these%20paths%20point%20to%20the%20correct%20CUDA%20installation.))

ExplorerFzw, 2025-05-16:
Yes, I installed jax in a conda environment using pip install -- upgrade "jax [kuda12]". I tried deleting LD_LIBRRY-PATH from the bashrc file, but it didn't work. The result of nvcc -- version is
///
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Wed_Aug_14_10:14:07_PDT_2024
Cuda compilation tools, release 12.6, V12.6.68
Build cuda_12.6.r12.6/compiler.34714021_0
///

chungmin99, 2025-05-19:
Hi! (also @brentyi for reference)

> Furthermore, in my understanding, the current issue may be caused by compatibility issues between jaxls and cusolver or some other problem.

As a sanity check, do either of these run on your machine?

1.  a snippit from the issue @chenyenru linked:
```
a = jax.numpy.array([[ 1.01290589e-03,  2.75272126e-05, -2.69166597e-04,
        -5.58780779e-06],
       [ 2.75272126e-05,  1.34740128e-03, -4.34192721e-06,
        -3.00849575e-04],
       [-2.69166597e-04, -4.34192721e-06,  1.28766222e-04,
         7.41944929e-07],
       [-5.58780779e-06, -3.00849575e-04,  7.41944929e-07,
         7.99537441e-05]])

numpy.linalg.cholesky(a)   # ok

jax.numpy.linalg.cholesky(a)   # NOk
```

2. [example code from jaxls](https://github.com/brentyi/jaxls/blob/main/examples/0_pose_graph.py)

---

One quick thing to try might be to change the solver backend to conjugate gradient:
```
sol = (
    jaxls.LeastSquaresProblem(factors, [joint_var])
    .analyze()
    .solve(
        verbose=False,
        linear_solver="conjugate_gradient",    # from "dense_cholesky"
        trust_region=jaxls.TrustRegionConfig(lambda_initial=1.0),
    )
)
```
I'm not sure if this will work, but just in case it's just the cholesky decomposition that's broken.

chungmin99, 2025-06-03:
Flagging a relevant issue -- another thing you could try:

> I ran into a similar error because of this issue: https://github.com/jax-ml/jax/issues/29042
> 
> You could try upgrading `nvidia-cublas-cu12` manually, or install `jax==0.6.0[cuda12]` in a new environment. 

 _Originally posted by @brentyi in [#14](https://github.com/chungmin99/pyroki/issues/14#issuecomment-2932926357)_

chungmin99, 2025-07-18:
I'll close this issue for now, but please feel free to re-open if there are still problems or other questions!