chungmin99/pyroki #13: Debug in JAX

Noietch, 2025-05-16:
Thank you for your work. This codebase seems much more lightweight than curobo. As I was carefully reading the code, I got stuck debugging the JAX code. Could you please share how you conduct debugging during development? Due to JIT, I am unable to inspect specific numerical values, even after using jax.config.update("jax_disable_jit", True).

chungmin99, 2025-05-19:
Hi!

> Could you please share how you conduct debugging during development?

Personally, at least for pyroki, I roughly follow these steps when implementing a new cost (e.g., collision costs):
1. write core components e.g., [`collide`](https://github.com/chungmin99/pyroki/blob/769c3f78d93934285d08a538dbcfc2010768f5de/src/pyroki/collision/_collision.py#L55)
2. write the cost function e.g., [`self_collision_cost`](https://github.com/chungmin99/pyroki/blob/769c3f78d93934285d08a538dbcfc2010768f5de/src/pyroki/costs/_costs.py#L170)
3. plug into the optimizer
    - if I run into NaNs here, I check for numerical instabilities e.g., through div-by-zeros

and I try my best to check in (1, 2) that the values are as expected.

That said, I definitely understand that JAX can be a pain for debugging -- I'm also fairly new to it so I'm not super sure about the best practices.

> even after using jax.config.update("jax_disable_jit", True).

Yea ... the optimizer (`jaxls`) heavily uses JAX transformations (e.g., `vmap`) and those will always return tracers, this is a minimal code snippit to see the tracer / JIT behavior:

```
import jax
import jax.numpy as jnp

DISABLE_JIT = True

if DISABLE_JIT:
    import os
    os.environ["JAX_DISABLE_JIT"] = "True"
    jax.config.update("jax_disable_jit", True)

def foo(x):
    # Print value -- may be a tracer. For first call it should print something.
    print(x)

    # Shouldn't work with tracers.
    if jnp.isnan(x).any():
        print("foo")
    return x

foo(1) # works always.

# fails normally, succeeds with DISABLE_JIT=True
jax.jit(foo)(1)

# fails regardless of DISABLE_JIT
jax.vmap(foo)(jnp.array([1, 2, 3]))
```

Noietch, 2025-05-19:
Thank you for your prompt reply! You mentioned that your code heavily relies on the `vmap` feature. In that case, why do you need to consider the batch dimension within the forward kinematics? As far as I know, JAX's `vmap` doesn't require explicit consideration of the batch in the code.

closed 2025-05-21 by Noietch