chungmin99/pyroki #14: Example execution error

robokang, 2025-05-16:
Dear author,

When I run the example, I get the following error message. 
If you can tell me how to fix it, I would appreciate it.

```
server-0:~/local/src/pyroki/examples$ python 01_basic_ik.py 
╭──────────────── viser ────────────────╮
│             ╷                         │
│   HTTP      │ http://localhost:8080   │
│   Websocket │ ws://localhost:8080     │
│             ╵                         │
╰───────────────────────────────────────╯
2025-05-17 05:14:53.300 | INFO     | jaxls._core:analyze:104 - Building optimization problem with 2 costs and 1 variables.
2025-05-17 05:14:53.318 | INFO     | jaxls._core:analyze:198 - Vectorizing group with 1 costs, 1 variables each: limit_cost
2025-05-17 05:14:53.374 | INFO     | jaxls._core:analyze:198 - Vectorizing group with 1 costs, 1 variables each: _pose_cost_analytical_jac
Traceback (most recent call last):
  File "/home/server-0/pyroki/examples/01_basic_ik.py", line 57, in <module>
    main()
  File "/home/server-0/pyroki/examples/01_basic_ik.py", line 41, in main
    solution = pks.solve_ik(
               ^^^^^^^^^^^^^
  File "/home/server-0/pyroki/examples/pyroki_snippets/_solve_ik.py", line 34, in solve_ik
    cfg = _solve_ik_jax(
          ^^^^^^^^^^^^^^
jaxlib.xla_extension.XlaRuntimeError: INTERNAL: cuSolver internal error
--------------------
For simplicity, JAX has removed its internal frames from the traceback of the following exception. Set JAX_TRACEBACK_FILTERING=off to include these.
```

chungmin99, 2025-05-19:
It seems that this is installed with the CUDA version of jax, and is similar to #12 .

The quickest fix would be using the CPU version (https://docs.jax.dev/en/latest/installation.html)...

Can you share more details on the system you're using (e.g., the JAX / CUDA version)?

brentyi, 2025-06-02:
I ran into a similar error because of this issue: https://github.com/jax-ml/jax/issues/29042

You could try upgrading `nvidia-cublas-cu12` manually, or install `jax==0.6.0[cuda12]` in a new environment.