chungmin99/pyroki #20: Multiple IK target with floating base robot

YoontaeCho, 2025-05-19:
Thank you for your awesome work!
In your project demo video, there's a G1 humanoid that achieves multiple IK target(E.g. feet and hands)

[image]

I tried to implement this demo, but I found that the thing is not very simple. 
The pre-defined cost function which supports multiple IK target: `pose_cost_analytic_jac` does not support floating base, while the cost function with floating base: `pose_cost_with_base` does not support multiple IK target. 

Would you mind to upload the code that used to make your demo video, in the `examples`? 
It would be very helpful me to write cost function for the floating base robots.

Best regards, Yoontae

chungmin99, 2025-05-19:
Hi!

The demo can be found here: https://gist.github.com/chungmin99/c644aa58cadb5932404b13bbe9629a21

This code should be similar to the [existing floating base example](https://github.com/chungmin99/pyroki/blob/main/examples/pyroki_snippets/_solve_ik_with_base.py), but with slight modifications on the base pose handling (to initialize and bias towards the previous pose, not identity).

(I guess I didn't include it in the official list of examples...)

One minor comment:
> `pose_cost_with_base` does not support multiple IK target.

I want to point out that the above code _does_ use `pose_cost_with_base` for multiple IK targets -- these costs should (generally) work for an arbitrary number of joint targets. For example, the pose cost is shared between the [single-target](https://github.com/chungmin99/pyroki/blob/3464ecdb71c2f6a3c1582bc86b4e2ff679c553a6/examples/pyroki_snippets/_solve_ik.py#L53-L61) and [bimanual](https://github.com/chungmin99/pyroki/blob/3464ecdb71c2f6a3c1582bc86b4e2ff679c553a6/examples/pyroki_snippets/_solve_ik_with_multiple_targets.py#L67C12-L74C11) examples.

YoontaeCho, 2025-05-19:
Thank you for your prompt response and code!

Yepp, I found that I misunderstood the code. Thank you for your reminder too.
One small followingup question: What's the difference of `pose_cost_analytic_jac` over `pose_cost`?

closed 2025-05-19 by YoontaeCho