chungmin99/pyroki #26: High GPU Memory Usage (~18GB) When Running Pyroki

caozx1110, 2025-05-20:
Hi, thank you for your great work on Pyroki!

I'm currently using Pyroki and noticed that it consumes around 18GB of GPU memory, which seems quite high. I'm wondering:

1. Is this level of memory usage expected?

2. What are the main factors contributing to such high GPU memory usage?

3. Are there any recommended ways to reduce the memory consumption, such as configuration options, model changes, or batch size adjustments?

Any guidance would be greatly appreciated. Thanks again for the excellent tool!

[image]

chungmin99, 2025-05-20:
Yes this memory allocation is expected (JAX preallocates memory, as documented [here](https://docs.jax.dev/en/latest/gpu_memory_allocation.html)) -- but you can disable it with `XLA_PYTHON_CLIENT_PREALLOCATE=false`.

caozx1110, 2025-05-21:
Thank you for the quick response!
It worked perfectly.
Really appreciate your help!

closed 2025-05-21 by caozx1110