chungmin99/pyroki #32: Does it work for humanoid robot w/ wheels?

jkim447, 2025-05-27:
Thanks for the great work, this looks super promising!

Would your code work for a wheel-based humanoid, like a Galaxea Robot?

For instance, I'd like to provide a set of goal poses and come up with the joints / controls to reach such target poses.

Just wanted to verify if this was possible with your code at all!

chungmin99, 2025-05-28:
The code should work for a wheel-based humanoid, similar to the mobile robot example [here](https://github.com/chungmin99/pyroki/blob/main/examples/03_mobile_ik.py); given a target pose (or multiple) we should be able to solve for the robot joint configurations.

The base pose would be provided as SE3 transformations -- if you wanted to get the actual controls for the motors you would need a higher level planner (~parallel parking a car), though I'm not very familiar with this literature.

jkim447, 2025-05-29:
Got it, thanks this is helpful!

closed 2025-05-29 by jkim447