chungmin99/pyroki #34: Performance Issue in `10_humanoid_retargeting.py`: Extremely Long Exit Time in `solve_retargeting` Suspected Related to JAX JIT

shieldkeke, 2025-05-29:
Hi! I am experiencing severe performance issues when running `10_humanoid_retargeting.py`. The details are:

- Tested on Python 3.10, 3.12, and 3.13, and the issue persists across all versions.
- When the `solve_retargeting` function is decorated with `@jax.jit`:
  - It takes only a few seconds from entering the function to obtaining the solution.
  - However, the function exit (the `return` statement) takes an extremely long time, ranging from tens of seconds to **over a minute**.
- When `@jax.jit` is removed:
  - Obtaining the solution itself takes tens of seconds to over a minute.
- Modifying the function to accept a list containing multiple trajectories causes the function to hang and **never exit**.

I am using the CUDA 12 enabled JAX version (`jax[cuda12]`) on an NVIDIA RTX 4080s GPU.

When testing on Python 3.10, I encountered the following warning in the console:

```
W external/xla/xla/service/gpu/autotuning/dot_search_space.cc:200] All configs were filtered out because none of them sufficiently match the hints. Maybe the hints set does not contain a good representative set of valid configs? Working around this by using the full hints set instead.
```

When processing a list of trajectories, I encountered the following warning:
```
E external/xla/xla/service/slow operation alarm.cc:73].  
```
and
```
[Compiling module jit solve retargeting] very slow compile? If you want to file a bug, run with envvar XLA_FLAGs=--xla_dump_to=/tmp/foo and attach the results.
```

- Is this significant delay on function exit related to JAX’s JIT compilation or runtime behavior?
- Are there any known optimizations or workarounds to reduce the exit overhead?
- Is the issue with passing multiple trajectories known, and how can it be resolved?

Thank you!

shieldkeke, 2025-05-30:
Thank you very much for your detailed and thoughtful reply — I really appreciate your dedication to this work!

I roughly understand your explanation regarding the runtime behavior when using the `@jit` decorator. Inside the jit-decorated function, JAX might not wait for the operation to complete, so removing the outermost jit decorator allows measuring the true runtime, right?

When I tested retargeting on a trajectory of length 128 — without collision checking, but considering joint velocity constraints, joint smoothness, local and global alignment, scale regularization, and joint position limit costs — it took about 2 seconds for analysis and 32 seconds to solve. Honestly, that surprised me.

Regarding the warning below, I have only encountered it when running `10_humanoid_retargeting.py` or its modified versions, on Python 3.10, 3.12, and 3.13. Other robot arm examples and trajectory optimization tasks also aren’t very fast — at least not millisecond-scale, more like 10+ seconds — on my RTX 4080s machine. The `10_humanoid_retargeting.py` script takes tens of seconds. I wonder if you've seen this warning on your system and how fast each of these scripts can run?

```
W external/xla/xla/service/gpu/autotuning/dot_search_space.cc:200] All configs were filtered out because none of them sufficiently match the hints. Maybe the hints set does not contain a good representative set of valid configs? Working around this by using the full hints set instead.
```

About why I pass in a list rather than using `vmap`: I simply wanted to test whether accumulating n times the operations inside the jit-decorated `solve_retargeting` function would lead to roughly n times the exit time.  (Actually, I passed in a list of 18 trajectories, each containing 100+ time steps.) But the result was surprising — it never exited and produced an additional warning:

```
E external/xla/xla/service/slow operation alarm.cc:73].
[Compiling module jit solve retargeting] very slow compile? If you want to file a bug, run with envvar XLA_FLAGs=--xla_dump_to=/tmp/foo and attach the results.
```

I’m wondering if solving all problems exclusively via `jaxls.LeastSquaresProblem` inherently leads to slower results? It would be great if multiple solver options were available — for example, allowing just solving a **QP** problem — which I believe could be much faster.

Additionally, I seem unable to handle multiple trajectories with different lengths in a batch using `vmap`, so I can’t currently provide timing for a batched vmap approach. Implementing that might require significant changes.

matteodeval, 2025-07-04:
Hi guys, as pointed out also in the issue #41 , there is the same problem related to the first calculation time also in the `07_trajectory_optimisation `and also in the `06_online_planning`. I'm not working at the moment on the `10_humanoid_retargeting`, but I wouldn't be surprised if it has the same problematic.

> To avoid the overhead, perhaps caching is useful: https://docs.jax.dev/en/latest/persistent_compilation_cache.html

I tried to follow your advice @chungmin99 and it seems working on the `06_online_planning`! This is how i changed the beginning of the example code `06_online_planning`, by adding the part "import os + os.environ etc" and the part "import jax + jax.config etc..."

```
"""Online Planning

Run online planning in collision aware environments.
"""

import time

import os
os.environ["JAX_COMPILATION_CACHE_DIR"] = r"C:\Users\matte\Desktop\PyRoki\jax_cache"

import numpy as np
import pyroki as pk
import viser
from pyroki.collision import HalfSpace, RobotCollision, Sphere
from robot_descriptions.loaders.yourdfpy import load_robot_description
from viser.extras import ViserUrdf

import pyroki_snippets as pks

import jax
jax.config.update("jax_persistent_cache_min_entry_size_bytes", -1)
jax.config.update("jax_persistent_cache_min_compile_time_secs", 0)

def main():
```

I did the same thing on the python file `_online_planning `present in the pyroki_snippets folder

```
from typing import Sequence
import time

import os
os.environ["JAX_COMPILATION_CACHE_DIR"] = r"C:\Users\matte\Desktop\PyRoki\jax_cache"

import jax
import jax.numpy as jnp
import jax_dataclasses as jdc
import jaxlie
import jaxls
import numpy as onp
import pyroki as pk

jax.config.update("jax_persistent_cache_min_entry_size_bytes", -1)
jax.config.update("jax_persistent_cache_min_compile_time_secs", 0)
```

I did this copy paste in both of the folders because at the beginning I tried to modify just the python code `_online_planning`, but it wasn't saving any file in the cache folder. I also tried to modify just the python file `06_online_planning `example, and it wasn't saving anything in the folder cache. So for the moment i figured out that by modifying both of the python files is actually saving the cache informations inside the folder.

These are the results that I discovered until now:

1)  The compile process for XLA, without cache, a.k.a. initialise the function `_solve_online_planning_jax`, on my machine (laptop Legion 5 Pro 16ach6h, Ryzen 7 5800h, 16gb RAM, RTX 3070m 8gb) in the example file `06_online_planning`, it takes between 45 and 73 seconds. 45 seconds on Windows 11, 73 seconds on Linux

2) after the compile process for XLA is completed, the code run basically almost in real-time, the time printed (that before was 45 or 73 seconds as first "initialisation") drop to 0.2 to even 0.0000etc seconds. The simulation on Viser is basically real-time interactive

3) By adding this cache method, the initial compilig time drop drastically to 10-15 seconds. At the moment for me it's more than enough to implement the PyRoki system without having a bottle-neck of more than a minute even if I just want to add a print 👍 

4) I tested the cache method also by switching from a Franka Robotics Panda to a ABB CBR 15000: of course the new robot needs to be recompiled again, but then its cache is saved in the cache folder without overwrite the previous cache saved from the Franka robot. When I choose again the Franka Robot, it doesn't need to recompile again, it takes its 10 seconds.

5)  @shieldkeke as far as I understood, using CUDA enabled JAX it's useless in this case unfortunately

> I am using the CUDA 12 enabled JAX version (jax[cuda12]) on an NVIDIA RTX 4080s GPU. 

I started to use PyRoki on Windows, but then when i discovered that GPU accelleration was available just on Linux I moved there. Result: There is no difference having JAX running on GPU or on CPU for the compile process for XLA, because anyway the compile process will always work just on the CPU; the GPU it can be used for the post-compilation part, where you can choose GPU, TPU etc. But anyways, at the moment I didn't see such strong bootlenecks for the real-time part (so the "post-compilation part") by using just my CPU, I wouldn't notice any difference between my Ryzen 7 and my RTX 3070.

I hope this can help in some ways! I'll send new updates if I get other solutions