chungmin99/pyroki #47: Trouble when urdf file including floating joint

goldenstasteworriors, 2025-07-02:
@brentyi @chungmin99 @danielpmorton
 I failed to use it when my urdf file including floating joint.Error occurs:  File "/home/yekangjie/Downloads/project/pyroki/src/pyroki/_robot_urdf_parser.py", line 448, in _get_joint_limits
    raise ValueError(f"Joint '{joint.name}' ({joint.type}) has no limits.")
ValueError: Joint 'floating_base_joint' (floating) has no limits.
I don't know how to set limit on floating joints.How to fix this problem?

chungmin99, 2025-08-01:
Hi, sorry for the delay!

> I don't know how to set limit on floating joints.How to fix this problem?

We don't currently support URDFs with floating joints... :( What type of robot are you trying to load? 

If it's a mobile robot (with one floating joint between itself and the world), perhaps you could
1) remove the floating joint from the URDF, and
2) refer to the approach in [the mobile robot IK example](https://github.com/chungmin99/pyroki/blob/main/examples/03_mobile_ik.py) -- this jointly solves for the base pose + robot joint configuration.

chaitanya1chawla, 2025-12-03:
On the other hand [if its not a wheeled robot, but legged], you can add the base joint also as one of the optimizing variables in jaxls.
I have this and a locomotion policy on top to follow different base poses.